jacopotediosi/OctoPrint-Telegram #386: Reprint item in menu

GeralddenEngelsman, 2021-12-08:
Is it possible to add a /reprint option that does not require you to have all the steps to go through  file selection etc. On Octoprint, you can simply press print again, as the file is still "loaded" while you did not reboot/change to another file.

giloser, 2021-12-28:
Hi,
I have prepare an /again that will do something similar on the next release.

GeralddenEngelsman, 2021-12-28:
> Hi, I have prepare an /again that will do something similar on the next release.

Awesome! If you want me to help test it, let me know!

zaprodk, 2025-03-06:
We're in 2025 and the function is still not there LOL :D

jacopotediosi, 2025-07-08:
There is no need for a separate command.
The commit above makes `/print` offer to print the already loaded file, if present. Since in OctoPrint the file remains loaded after a print finishes, simply using the `/print` command will print it again without needing to search for it in the file list.

This change will be released in the next version.

Thank you for your report.

commit d4739266 (2025-07-08): Offer previously loaded file in /print

Close #386
closed 2025-07-18 by jacopotediosi