jacopotediosi/OctoPrint-Telegram #387: Octo4a gif cpulimit

MischaI, 2021-12-12:
On octo4a(https://github.com/feelfreelinux/octo4a) cant Create gif, cause cpulimit not installed.
How I can install cpulimit on Android?
Or another Solution to Create gifs and Send via telegram?

giloser, 2021-12-28:
Hi sorry for the delay.
you could try something like this https://github.com/Miliox/android-cpulimit

I don't have a plan for other solution for now the only thing I could think of is to put an option to remove the use of cpulimit but that will be at your own risk as creating the gif could be fatal to print on some device.

MischaI, 2021-12-28:
Yes, I See, but some android Devices are strong enouth, I think, and I can test this change (without cpulimit), cause my s10 hasnt root, and display broken, so I cant root s10.

MischaI, 2021-12-28:
Can you make is as Checkbox or Option? 
User should check or uncheck, depend of device.

commit 56b9f7d5 (2025-08-20): Allow cpulimit bypass

Fix #387, fix #382