jacopotediosi/OctoPrint-Telegram #389: Error: [ERR GET IMAGE]

Ash3781, 2021-12-30:
I get the error [ERR GET IMAGE] in each message from the bot, everything else works but not the image, i've tried setting the snapshot address, but it still does not work: i'll add a picture of the settings
[image]

Ash3781, 2021-12-30:
after a error i restarted the whole pi and the printer, and after that nothing worked anymore, so i take that "everything else works" back
[image]

giloser, 2021-12-30:
Hi ,

for the image if it doesn't work on octoprint it will not work in telegram. So first you have to made it work in test snapshot in octoprint.

for the rest send me a log so I can take a look at what is happening please.

Ash3781, 2021-12-30:
sorry for the late reply, i just finished setting it up, i just flashed the normal octopi software onto it, instead of installing it on top of debian, it works for now, tho i need to set up the second cam and i'll probably have more question once both a re running, i'll keep you updated

Ash3781, 2021-12-30:
apparently everything works (for now), still thanks^^

closed 2021-12-30 by Ash3781