jacopotediosi/OctoPrint-Telegram #392: too many "wrong" messages

RoboCop0815, 2022-01-25:
I use in end G-code dir M300 Sxxx Pxxx to play a melody via the beeper (31 lines).
Since the last updates I get on Telegram as many messages as M300 parts in G-code with the following message:
 "🔔 Printer received alert command Bed 57.8/0.0, Extruder 180.5/0.0".
What is this related to and how to change it.
Here is the code.
...
M300 S2093 P214
M300 S932 P214
M300 S2093 P214
M300 S1567 P214
M300 S1244 P214
M300 S1567 P214
M300 S1046 P428
M300 S2093 P214
M300 S932 P214
M300 S2093 P214
M300 S1567 P214
M300 S1244 P214
M300 S1567 P214
M300 S1046 P428
M300 S2093 P214
M300 S2349 P214
M300 S2489 P214
M300 S2093 P107
M300 S2489 P214
M300 S2093 P107
M300 S2489 P214
M300 S2349 P214
M300 S932 P107
M300 S2349 P214
M300 S932 P107
M300 S2349 P214
M300 S2093 P214
M300 S932 P214
M300 S1567 P214
M300 S932 P214
M300 S2093 P428

Translated with www.DeepL.com/Translator (free version)

RoboCop0815, 2022-01-26:
I already had this idea, but how do I get real alerts?

giloser, 2022-01-26:
for the printer they are all real alert

jacopotediosi, 2025-06-24:
Receiving notifications for alerts is a feature of this plugin.  
You’re receiving these messages because in the plugin settings you checked the *Alert* checkbox for your chat.  
If you want to disable them, go to *OctoPrint Settings → Telegram → Known Chats*, click the *Notify* icon in the row of your chat, and uncheck *Get Notification at... Alert*.