jacopotediosi/OctoPrint-Telegram #393: Many messages at M600

DerT94, 2022-01-25:
I use the M600 command when my filament runout sensor triggerd.
In this moment the telegram adapter starts to spam me.

[image]

Deses, 2022-07-17:
Did your printer resume printing? Mine didn't. Please see https://community.octoprint.org/t/change-filament-command-m600-doesnt-resume-printing-if-telegram-plugin-is-installed/45538

jacopotediosi, 2025-08-21:
This can happen if your printer sends additional text on the serial port between one `echo:busy: paused for user` message and the next.
I’ll look into it and try to find a solution.  
It would be helpful to have a snippet of your serial output while this issue occurs.

commit 697bcc33 (2025-08-22): Fix repeated plugin_pause_for_user_event_notify notifications

Fix #393