jacopotediosi/OctoPrint-Telegram #394: Not working?

ik2wxx, 2022-02-10:
Hi all
I've installed the plugin by the plugin manager using url, the plugin install but it's disabled. 
So I enabled it, but no "telegram" menu option shows up.
I restarted octopi and the plugin appears as disabled.
I'm using last version of octopi.
Any help?
Tnx

mkrasselt1, 2022-03-29:
i am having kind of the same problem, after installing/updating, it says "unknown plugin" installed and it shows up nowhere - can i attach some log for debugging ?

jginesclavero, 2022-05-07:
I am having the same problem...

mkrasselt1, 2022-07-27:
> Same here

I only managed to solve it by reflashing the octoprint image. May be related to mine beeing very old installation only beeing updated over the jears. Did an backup and reinstalled everything.

(tryed debugging and it has something to do with the image library failing to start, but I couldn't resolve the issue no matter what I tried to repair and reinstall)

svcru, 2022-07-31:
I installed a fresh system 2 weeks ago.
And I solved the problem, you need to install the **libopenjp2-7 package**
```
sudo apt-get install libopenjp2-7
```

mkrasselt1, 2022-07-31:
great, tried it at my old backup, would work again :)) isnt it installed by default anymore or a missing dependency ?

svcru, 2022-08-06:
> great, tried it at my old backup, would work again :)) isnt it installed by default anymore or a missing dependency ?

I'm new. I installed Octoprint for the first time just a month ago.
As I understand it, pip packages have dependencies, but apt packages do not have dependencies.