jacopotediosi/OctoPrint-Telegram #395: Not receiving status message

Myzhar, 2022-02-11:
With the latest version I'm not receiving the status message. I receive all the notifications, but not this.

What info do you need to detect the cause of the problem?

INeDiA, 2022-03-14:
> I guess this [line ](https://github.com/fabianonline/OctoPrint-Telegram/blob/eb939535ed1db861d2085364d3c4a31e7b1c1763/octoprint_telegram/__init__.py#L1988) is cause of problem. Need to change "URL" to "snapshot"
> 
> ```python
>     self._logger.debug(
>         "multicam profile:  " + str(li)
>     )
>     snapshot_url = li.get("snapshot")
>     self._logger.debug(
>         "multicam url :  " + str(snapshot_url)
>     )
> ```

I tried with this (and of course restarted thw whole thing) but nothing changed, I still can't receive status updates unless I manually ask for them

jacopotediosi, 2025-06-26:
Are you still experiencing this with the [latest version 1.7.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0)?

closed 2025-07-19 by jacopotediosi