jacopotediosi/OctoPrint-Telegram #398: GIF is still a picture

CommanderRedYT, 2022-03-13:
When entering the /gif command (or /supergif), i only get a single frame of it. It needs a bit of time so it probably gets all 30 / 60 frames but after that, i get just one picture. 

Also, the normal "Sending a video..."-Status in telegram is not showing

CommanderRedYT, 2022-03-13:
Note: The printer is not in connected state!

jacopotediosi, 2025-07-02:
If it still happens with the latest [1.7.0 release](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0), could you please attach the generated gif here?

Are you viewing the GIF on an Apple device? Have you tried to see if the same GIF works on another device? It could be related to #406.

CommanderRedYT, 2025-07-02:
Hey, I don't even have the printer anymore or any octoprint deployed, my setup changed quite a bit in 3 years. Maybe someone else can replicate it if it still exists, otherwise I would just close the issue

closed 2025-07-08 by jacopotediosi