jacopotediosi/OctoPrint-Telegram #403: /gif not working

Opadera, 2022-05-27:
Hello,

i'm trying to create a .gif with the /gif command. I see the status "recording a video" and "sending a file" in telegram but no video is returned.
I'm running OctoPrint 1.8.1 on Octo4A

Here' the relevant log info 
`2022-05-27 06:40:09,337 - octoprint.plugins.telegram.listener - INFO - Got a command: '/gif' with parameter: '' in chat 623024455
2022-05-27 06:40:09,379 - octoprint.plugins.telegram.TCMD - INFO - Will try to create a gif
2022-05-27 06:40:09,947 - octoprint.plugins.telegram - ERROR - Caught an exception trying clean previous images : [Errno 2] No such file or directory: '/root/.octoprint/data/telegram/tmpgif/gif.mp4'
2022-05-27 06:40:09,948 - octoprint.plugins.telegram - INFO - test if nice exist
2022-05-27 06:40:09,950 - octoprint.plugins.telegram - INFO - test exist program 'nice', '--version'
2022-05-27 06:40:10,051 - octoprint.plugins.telegram - INFO - ret = 1
2022-05-27 06:40:10,052 - octoprint.plugins.telegram - INFO - test if cpulimit exist
2022-05-27 06:40:10,053 - octoprint.plugins.telegram - INFO - test exist program 'cpulimit', '--help'
2022-05-27 06:40:10,122 - octoprint.plugins.telegram - INFO - ret = 1
2022-05-27 06:40:10,123 - octoprint.plugins.telegram - INFO - test if ffmpeg exist
2022-05-27 06:40:10,124 - octoprint.plugins.telegram - INFO - test exist program 'ffmpeg', '-h'
2022-05-27 06:40:10,169 - octoprint.plugins.octoeverywhere - INFO - Main request http URL threw an exception: HTTPConnectionPool(host='192.168.178.23', port=5001): Max retries exceeded with url: /mjpeg (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f156f5f70>: Failed to establish a new connection: [Errno 110] Operation timed out'))
2022-05-27 06:40:10,172 - octoprint.plugins.octoeverywhere - ERROR - Main request failed and we have no more fallbacks. We DON'T have a main response.
2022-05-27 06:40:10,173 - octoprint.plugins.octoeverywhere - WARNING - Web Stream http [62]  failed to make http request. octoHttpResult was None; url:http://192.168.178.23:5001/mjpeg
2022-05-27 06:40:10,342 - octoprint.plugins.telegram - INFO - ret = 0
2022-05-27 06:40:11,170 - octoprint.plugins.telegram - INFO - sec=5
2022-05-27 06:40:11,172 - octoprint.plugins.telegram - INFO - timeSec=11:11:05
2022-05-27 06:40:11,174 - octoprint.plugins.telegram - INFO - limit_cpu=260.0 | used_cpu=4.0 | because nb_cpu=8
2022-05-27 06:40:11,222 - octoprint.plugins.telegram - INFO - will now create the video  'nice', '-n', '20', 'cpulimit', '-l', '260.0', '-f', '-z', '--', 'ffmpeg', '-y', '-threads', '4.0', '-i', 'http://192.168.178.11:1111/mjpeg', '-t', '11:11:05', '-pix_fmt', 'yuv420p', '/root/.octoprint/data/telegram/tmpgif/gif.mp4'
2022-05-27 06:40:13,165 - octoprint.plugins.telegram - INFO - Finish the video
2022-05-27 06:40:24,656 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [110] GET [upload:0.013s; request_exe:0.929s; compress:-1.000s send:0.041s] size:(84->84) compressed:False msgcount:1 type:application/json status:200 cached:False for http://127.0.0.1:5000/api/languages
2022-05-27 06:40:24,750 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [111] GET [upload:0.019s; request_exe:0.967s; compress:-1.000s send:0.039s] size:(199->199) compressed:False msgcount:1 type:application/json status:200 cached:False for http://127.0.0.1:5000/api/plugin/appkeys?all=true
2022-05-27 06:40:25,081 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [109] GET [upload:0.012s; request_exe:1.357s; compress:-1.000s send:0.055s] size:(0->0) compressed:True msgcount:1 type:None status:304 cached:False for http://127.0.0.1:5000/api/printerprofiles
2022-05-27 06:40:25,243 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [114] GET [upload:0.032s; request_exe:1.385s; compress:0.001s send:0.046s] size:(1247->519) compressed:True msgcount:1 type:text/html; charset=utf-8 status:200 cached:False for http://127.0.0.1:5000/api/plugin/telegram
2022-05-27 06:40:25,409 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [113] GET [upload:0.015s; request_exe:1.446s; compress:0.002s send:0.065s] size:(5219->1221) compressed:True msgcount:1 type:application/json status:200 cached:False for http://127.0.0.1:5000/plugin/logging/
2022-05-27 06:40:25,518 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [112] GET [upload:0.011s; request_exe:1.591s; compress:0.007s send:0.068s] size:(581->244) compressed:True msgcount:1 type:application/json status:200 cached:False for http://127.0.0.1:5000/plugin/backup/
2022-05-27 06:40:25,873 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [115] GET [upload:0.052s; request_exe:1.695s; compress:0.001s send:0.220s] size:(715->344) compressed:True msgcount:1 type:text/html; charset=utf-8 status:200 cached:False for http://127.0.0.1:5000/api/plugin/telegram?bindings=true
2022-05-27 06:40:25,981 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [117] GET [upload:0.062s; request_exe:1.426s; compress:-1.000s send:0.055s] size:(0->0) compressed:True msgcount:1 type:None status:304 cached:False for http://127.0.0.1:5000/plugin/pluginmanager/repository
2022-05-27 06:40:26,079 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [118] GET [upload:0.018s; request_exe:1.471s; compress:-1.000s send:0.059s] size:(0->0) compressed:True msgcount:1 type:None status:304 cached:False for http://127.0.0.1:5000/plugin/pluginmanager/orphans
2022-05-27 06:40:26,197 - octoprint.plugins.octoeverywhere - INFO - Web Stream http [116] POST [upload:0.031s; request_exe:1.734s; compress:-1.000s send:0.037s] size:(124->124) compressed:False msgcount:1 type:text/html; charset=utf-8 status:200 cached:False for http://127.0.0.1:5000/api/plugin/telegram
2022-05-27 06:40:33,828 - octoprint.plugins.octoeverywhere - INFO - Web Stream ws   [67] websocket closed after131.14507603645325 seconds
2022-05-27 06:40:33,850 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
2022-05-27 06:40:33,861 - websocket - ERROR - error from callback <function Client.__init__.<locals>.onClosed at 0x7f138f89d0>: onClosed() takes 1 positional argument but 3 were given
`

any ideas how I can solve this?

valentt, 2022-10-31:
I tested latest OctoPrint and /gif works for me without problem on 20 printers tested so far.

MischaI, 2022-11-16:
i know this problem is on octo4a

jacopotediosi, 2025-07-23:
Duplicate of #387.

It is now possible to generate GIFs on **Octo4a** by checking the checkbox **"Don't use cpulimit/limitcpu"** in  
**Telegram plugin settings → General settings tab → Gifs generation**.

This checkbox is implemented in the `devel` branch, which can be installed via the OctoPrint Plugin Manager using the following URL:

```
https://github.com/jacopotediosi/OctoPrint-Telegram/archive/master.zip
```

It will be included in the upcoming v1.9.0 release.

As shown in the image below, the `/gif` command works correctly on Octo4a 2.1.0.

[image]