jacopotediosi/OctoPrint-Telegram #407: Cannot get more than 2 termistor outputs

karabas2011, 2022-08-12:
I use printer with 4 extruders, I use two as common extruders and two as a custom heaters.
I try to add lines with tool2 to code but octoprint failed to boot.
            e1_temp = temps["tool0"]["actual"] if "tool0" in temps else 0.0
            e1_target = temps["tool0"]["target"] if "tool0" in temps else 0.0
            e2_temp = temps["tool1"]["actual"] if "tool1" in temps else 0.0
            e2_target = temps["tool1"]["target"] if "tool1" in temps else 0.0
            
            i add
            e3_temp = temps["tool2"]["actual"] if "tool2" in temps else 0.0
            e3_target = temps["tool2"]["target"] if "tool2" in temps else 0.0

jacopotediosi, 2025-07-02:
The variable shortcuts currently available in notifications do indeed support a maximum of two extruders:  
```
{bed_temp} {e1_temp} {e2_temp}	Current temperatures of bed / extruder 1 / extruder 2.
{bed_target} {e1_target} {e2_target}	Target temperatures of bed / extruder 1 / extruder 2.
```

However, as explained in the "variables" spoiler within plugin settings, you can use `{temps}` as a variable in the notification messages, which should include all extruders on your printer:

[image]

For example, try adding `{temps}` in a notification message to see its structure, or use it like this: `{temps[tool2][actual]}`.

jacopotediosi, 2025-07-02:
The above commit adds shortcuts for up to 5 extruders. It will be included in the next release.
Thanks for your report.

commit af9ebd8d (2025-07-02): Support up to 5 tools in notifications

Close #407
commit f802df0a (2025-07-18): Support up to 5 tools in notifications

Close #407
closed 2025-07-18 by jacopotediosi