jacopotediosi/OctoPrint-Telegram #410: Send notification every not working on OP1.8.6 plugin 1.6.5

Aliens82, 2022-12-18:
Hi all,

I can report issue on my Octoprint ver 1.8.6 plugin Telegram ver 1.6.5, when i setting send notification every any time or any height mm, i get a notification every Z change. If i setting 2mm, i get notification every 0.2mm, if print set is 0.2mm height. 
Finally, I want to thank you for a great plugin. Its amazing.
I hope I don't repeat the problem. But I couldn't find him anywhere.

roboraptor, 2022-12-21:
Hi, I suffer the same problem but never really had time to thinker around it. 
It worked fine last time I guess when i was on Python 2 if I remember correctly. 
After the update it seems like sending GIF only on every Z step.
I can fiddle with Zheight or MaxTime but nothing seems to change when i run another print.
I just muted the telegram chat and set autodelete on 1day but this way I cant get any notification on Errors of Fail Alerts

arossbach10, 2023-10-04:
I have the same issue. 
Telegram Plugin was working some time before. 
But now with an update to 1.6.5 I don't receive a notification. /status command does not send a response.
I also updated to python 3.

jacopotediosi, 2025-06-24:
I believe this was a duplicate of #13. If you still encounter this bug in [version 1.7.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0), please open a new issue and attach the logs, which can be found under *OctoPrint Settings → Logging → plugin_telegram.log*.

closed 2025-06-24 by jacopotediosi