jacopotediosi/OctoPrint-Telegram #411: Cant install inside a docker container

SpawW, 2022-12-21:
Enviroment: Raspberry Pi 4 / octoprint in container docker
The install ends with "success" but after restart the plugin is not available and the plugin installed pop-up appears a "unknow plugin installed".

"Plugin management log

    Uninstall Bed Visualizer:
    Install Unknown: 

A restart is needed for the changes to take effect.
"

I make a test for install/uninstall "Bed Visualizer" plugin and works fine in same container.
"

logs:
----------------

Installing plugin "OctoPrint-Telegram" from https://github.com/fabianonline/OctoPrint-Telegram/archive/master.zip.../usr/local/bin/python -m pip --disable-pip-version-check install file:///tmp/tmprf40gwu3/OctoPrint-Telegram-master.zip --no-cache-dirProcessing /tmp/tmprf40gwu3/OctoPrint-Telegram-master.zipPreparing metadata (setup.py): startedPreparing metadata (setup.py): finished with status 'done'Requirement already satisfied: OctoPrint in /octoprint/plugins/lib/python3.8/site-packages (from OctoPrint-Telegram==1.6.5) (1.8.6)Requirement already satisfied: pillow in /octoprint/plugins/lib/python3.8/site-packages (from OctoPrint-Telegram==1.6.5) (9.3.0)Requirement already satisfied: requests[socks] in /usr/local/lib/python3.8/site-packages (from OctoPrint-Telegram==1.6.5) (2.28.0)Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (0.8.0)Requirement already satisfied: werkzeug<2.1,>=2.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.0.3)Requirement already satisfied: emoji<2,>=1.4.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.7.0)Requirement already satisfied: OctoPrint-PiSupport>=2022.3.28 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2022.6.13)Requirement already satisfied: pathvalidate<3,>=2.4.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.5.0)Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.8.3)Requirement already satisfied: Flask-Limiter<3,>=2.6 in /octoprint/plugins/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.6.3)Requirement already satisfied: watchdog<2,>=1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.0.2)Requirement already satisfied: pylru<2,>=1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.2.1)Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.0)Requirement already satisfied: filetype<2,>=1.0.7 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.0.13)Requirement already satisfied: pyserial<4,>=3.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (3.5)Requirement already satisfied: Click<9,>=8.0.3 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (8.1.3)Requirement already satisfied: unidecode in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.3.4)Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2021.2.23)Requirement already satisfied: feedparser<7,>=6.0.8 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (6.0.10)Requirement already satisfied: psutil<6,>=5.8 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (5.9.1)Requirement already satisfied: markdown<4,>=3.2.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (3.3.7)Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /octoprint/plugins/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (21.3.0)Requirement already satisfied: semantic-version<3,>=2.8.5 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.10.0)Requirement already satisfied: frozendict<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.3.2)Requirement already satisfied: future<1,>=0.18.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (0.18.2)Requirement already satisfied: websocket-client<2,>=1.2.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.3.2)Requirement already satisfied: wrapt<1.14,>=1.13.3 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.13.3)Requirement already satisfied: regex in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2022.6.2)Requirement already satisfied: blinker<2,>=1.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.4)Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (0.5.0)Requirement already satisfied: cachelib<0.3,>=0.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (0.2.0)Requirement already satisfied: tornado<7,>=6.0.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (6.1)Requirement already satisfied: colorlog<7,>=6 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (6.6.0)Requirement already satisfied: flask<2.2,>=2.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.1.2)Requirement already satisfied: Flask-Babel<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2.0.0)Requirement already satisfied: zeroconf<0.34,>=0.33 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (0.33.4)Requirement already satisfied: netifaces<1,>=0.11 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (0.11.0)Requirement already satisfied: sarge==0.1.6 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (0.1.6)Requirement already satisfied: zipstream-ng<2.0.0,>=1.3.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.3.4)Requirement already satisfied: sentry-sdk<2,>=1.5.7 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.5.12)Requirement already satisfied: PyYAML<6,>=5.4.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (5.4.1)Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (2021.10.11)Requirement already satisfied: passlib<2,>=1.7.4 in /octoprint/plugins/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Telegram==1.6.5) (1.7.4)Requirement already satisfied: idna<4,>=2.5 in /usr/local/lib/python3.8/site-packages (from requests[socks]->OctoPrint-Telegram==1.6.5) (3.3)Requirement already satisfied: charset-normalizer~=2.0.0 in /usr/local/lib/python3.8/site-packages (from requests[socks]->OctoPrint-Telegram==1.6.5) (2.0.12)Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/local/lib/python3.8/site-packages (from requests[socks]->OctoPrint-Telegram==1.6.5) (1.26.9)Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.8/site-packages (from requests[socks]->OctoPrint-Telegram==1.6.5) (2022.6.15)Requirement already satisfied: PySocks!=1.5.7,>=1.5.6 in /octoprint/plugins/lib/python3.8/site-packages (from requests[socks]->OctoPrint-Telegram==1.6.5) (1.7.1)Requirement already satisfied: argon2-cffi-bindings in /octoprint/plugins/lib/python3.8/site-packages (from argon2-cffi<22,>=21.3.0->OctoPrint->OctoPrint-Telegram==1.6.5) (21.2.0)Requirement already satisfied: sgmllib3k in /usr/local/lib/python3.8/site-packages (from feedparser<7,>=6.0.8->OctoPrint->OctoPrint-Telegram==1.6.5) (1.0.0)Requirement already satisfied: itsdangerous>=2.0 in /usr/local/lib/python3.8/site-packages (from flask<2.2,>=2.1->OctoPrint->OctoPrint-Telegram==1.6.5) (2.1.2)Requirement already satisfied: importlib-metadata>=3.6.0 in /usr/local/lib/python3.8/site-packages (from flask<2.2,>=2.1->OctoPrint->OctoPrint-Telegram==1.6.5) (4.11.4)Requirement already satisfied: Jinja2 [... cut ...] stalling collected packages: OctoPrint-TelegramAttempting uninstall: OctoPrint-TelegramFound existing installation: OctoPrint-Telegram 1.6.5Uninstalling OctoPrint-Telegram-1.6.5:Successfully uninstalled OctoPrint-Telegram-1.6.5WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venvSuccessfully installed OctoPrint-Telegram-1.6.5Done!

SpawW, 2022-12-21:
[image]
Screenshot of results (look during the install process appears a previous installation of plugin but is not available).

MrSoichi, 2024-03-31:
I have pretty much the same problem.  The plugin is installed, the token verification works, but the bot itself does not respond in any way to the start, status, etc. commands. How can I fix the warning about the root? Perhaps this will solve the problem?

closed 2025-06-26 by jacopotediosi