jacopotediosi/OctoPrint-Telegram #416: /status not working

ppetrov-dev, 2023-10-05:
I have no any messages on /status command sent
[image]

nitzerebbs, 2023-10-05:
same issue here, any solution for this ?

Sle77en, 2023-10-05:
Got the same issue. I do not receive information and if i send commands, nothing happens. API Token is valid.

nitzerebbs, 2023-10-05:
Creating a new bot does not help also

giloser, 2023-10-05:
could you send some log? I don't have lots of time for the moment but I'll try to take a look.

Sle77en, 2023-10-06:
Which log do you need? 
I also deleted the plugin and install it fresh. Does not help!

Sle77en, 2023-10-06:
> could you send some log? I don't have lots of time for the moment but I'll try to take a look.

[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/12830976/octoprint.log)

I tried different commands. Only /status is not working. And receiving print information doesnt work too.

majestix508, 2023-10-11:
After playing around I got a workaround:
in __init__.py I changed all "webcam" settings to "plugins", "classicwebcam" - cause the webcam is now a plugin, but that did not send an image but at least a notification with [ERR GET IMAGE].
When I set horizontal flip to true and save the settings in classicwebcam, then I get a notification with image.
the weird thing is that I get the url "DEBUG - Snapshot URL: http://127.0.0.1:1111/?action=snapshot" in octoprint.log, which is from the settings, but this url is not in my config.yaml

ppetrov-dev, 2023-10-21:
> weird thing

WORKS!!! Thanks a lot, man!

db0nd, 2023-11-11:
https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/master.zip 

Not work!

giloser, 2023-11-11:
hi!

yes I changed the GitHub to try to be able to update the main hub but
didn't get time to finish

try with the
https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/old_master.zip

I'll make the version work but looking at how people answer , I think I'll
stop after that.

db0nd, 2023-11-11:
> hi! yes I changed the GitHub to try to be able to update the main hub but didn't get time to finish try with the https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/old_master.zip I'll [...]

Thank you, it works!

lethuer, 2024-01-13:
> Hello! Is there any way this fix can be added to the main so it can be automatically be pulled in by octoprint?

I also would appreciate this !

wasalm, 2024-02-10:
I tried https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/old_master.zip and it looks like it is working.
Please add this to the main repo.

lethuer, 2025-05-12:
I just updated to 1.6.7 and it looks like it is still not merged ?

jacopotediosi, 2025-06-26:
Also, @lethuer are you using the OctoPi new camera stack?

lethuer, 2025-06-26:
Here is the log: [plugin_telegram.log](https://github.com/user-attachments/files/20930364/plugin_telegram.log)

with Telegram 1.7.0 installed at:
18:31 Octoprint comes online after installing 1.7.0
18:33 /status was requested
18:33 /settings were requested
18:34 printer connected
18:34 print was started

lethuer, 2025-06-26:
> can you share also your Classic Webcam / Multicam configurations?

Where to find them ?

lethuer, 2025-06-27:
I think it was already set to "DEBUG" in the previous log.
[image]

But anyhow, here is another one:
[plugin_telegram 20250627.log](https://github.com/user-attachments/files/20952550/plugin_telegram.20250627.log)

18:43 Restart because 1.7.0 was installed
18:44 Online message "Hello. I'm online and ready to receive your commands"
18:45 Online message "Hello. I'm online and ready to receive your commands"
18:46 /status
18:46 Online message "Hello. I'm online and ready to receive your commands"
18:47 /status
18:47 Online message "Hello. I'm online and ready to receive your commands"

After that I deleted the log, restarted OctoPrint again and repeated.
Here is the log from that one:
[plugin_telegram 20250627_2.log](https://github.com/user-attachments/files/20952574/plugin_telegram.20250627_2.log)

18:49 Goodbye message
18:49 Online message "Hello. I'm online and ready to receive your commands"
18:50 /status
18:50 Online message "Hello. I'm online and ready to receive your commands"

jacopotediosi, 2025-06-27:
@lethuer I think you have set the wrong snapshot URL in the Classic Webcam plugin.

Your current config is:
`'snapshot': 'http://localhost:8080/?action=snapshot', 'stream': '/webcam/?action=stream'`

Try going to OctoPrint Settings -> Classic Webcam -> set the Snapshot URL to `http://localhost/webcam/?action=snapshot` or `/webcam/?action=snapshot`.  With the latter it is possible that the Classic Webcam test button fails but octoprint-telegram works.

lethuer, 2025-06-28:
I replaced

Stream URL was:
/webcam/?action=stream

Snapshot URL was:
http://localhost:8080/?action=snapshot

I changed snapshot url to:
http://localhost/webcam/?action=snapshot

With that snapshot Test Button still works but /status does not work !

jacopotediosi, 2025-06-30:
However, it is not correct that failure to retrieve the image causes the entire message to fail, and this is definitely something I will address in the next release.

lethuer, 2025-06-30:
I'm not sure if that is helpful but I just want to let you know...
I use Logitech C270 Webcam.

> The image that is taken by the plugin is the same one you see when you press the Test button

Right now you did only mentioned the "Snapshot URL"...
But because the image is shown with the test button I assume that also the "Stream URL" I set must be okay ?

jacopotediosi, 2025-06-30:
Stream url is used for GIFs, snapshot url is used for photos. They have separate test buttons in classic webcam. Gifs need to check a specific checkbox in telegram plugin settings.

lethuer, 2025-06-30:
> They have separate test buttons in classic webcam.

Just for completeness: both test buttons are working...

lethuer, 2025-07-20:
I installed 1.8.0, restarted OctoPrint and verified the log level.

[image]

In the log I sent "/status" three times:
- 10:54 "/status" with response "Hello. I'm onlineand ready to receive your commands" (like before)

Then I tried to restart OctoPrint again what didn't work well (I hit the restart button twice)
- 10:56 Performing command for core:restart: sudo service octoprint restart (from "octoprint.log")

Then I disconnected power supply and connected again.
- 10:57 "I tried to send you a message, but an exception occurred."
- 10:58 "Hello. I'm online and ready to receive your commands."
- 10:59 "Hello. I'm online and ready to receive your commands." (why this is sent again ?!)
- 11:01 "/status" with image as response
- 11:01 "/status" with image as response

Then I restarted OctoPrit again (this time only hit the button once):
- 11:16 "Shuuting down. Goodbye"
- 11:17 "Hello. I'm online and ready to receive your commands."
- 11:17 "/status" with image as response

[20250720_plugin_telegram_1.8.0.log](https://github.com/user-attachments/files/21334027/20250720_plugin_telegram_1.8.0.log)

For me this os looking fine now ?

lethuer, 2025-07-20:
>> A fix has been submitted upstream to OctoPi just a few days ago. If you want to fix it in your environment, you'll find instructions here

I think the fix will come with an Update automatically in some time ?

jacopotediosi, 2025-07-20:
> I think the fix will come with an Update automatically in some time ?

Unfortunately not, OctoPi only updates by reflashing the entire SD card. In the link I provided, however, the instructions to follow are quite simple.

jacopotediosi, 2025-07-22:
@lethuer everything looks fine. Your log still contains an "exception in get_layer_progress_values", but it's a known bug that doesn’t cause any issues and will be fixed soon.
I'm glad we were able to resolve this issue.

closed 2025-07-22 by jacopotediosi