jacopotediosi/OctoPrint-Telegram #417: Not compatible with Octoprint 1.9

Jon-So, 2023-10-05:
Was working fine before I installed a new SD card, Token is Valid but the Bot wont connect... I think with the new camera stack might be a problem as well...

gary325, 2023-11-01:
Working OK with 1.9.3

Hardbaer, 2023-11-09:
It seems that it does not work with the changed camera stack in 1.9. Notifications without image work like before. As soon as an image is part of the notification it does not work anymore. The notification does not show up. Removing image in settings and the notification shows up again.

Hardbaer, 2023-11-09:
Came across #416. The solution solves at least the problem with the new camera stack.

jacopotediosi, 2025-07-13:
I just tested an installation of version 1.7.0 on OctoPi 1.0.0cam with the new camera stack. Everything worked out of the box. I only had to install `cpulimit` to get GIF generation working.

closed 2025-07-13 by jacopotediosi