jacopotediosi/OctoPrint-Telegram #419: Messages with images do not come through.

dylankird, 2023-11-02:
Any time I am supposed to receive a message with an image from the server (such as after sending /status) the response message is never received. It works fine when I turn off the mjpg-streamer server and I get the message with [ERR GET IMAGE], but when the mjpg-streamer server is on, I get nothing. Messages that I have set to not include an image (like the startup message) work fine regardless.

I am running Octoprint 1.9.3 with Telegram Notifications 1.6.5 on Debian : `Linux rigidbot-octoprint 6.1.0-13-686-pae #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) i686 GNU/Linux`

My webcam is set up as a Classic Webcam with the following settings:
[image]

I looked at the error logs after sending /status and this is what is there:

```
2023-11-01 20:23:15,854 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 86204352
2023-11-01 20:23:15,860 - octoprint.plugins.telegram - DEBUG - self.tmsg = <octoprint_telegram.telegramNotifications.TMSG object at 0xaf158910>
2023-11-01 20:23:15,862 - octoprint.plugins.telegram - DEBUG - Got an event: StatusNotPrinting Payload: {}
2023-11-01 20:23:15,865 - octoprint.plugins.telegram.TMSG - DEBUG - event :StatusNotPrinting
2023-11-01 20:23:15,876 - octoprint.plugins.telegram.TMSG - DEBUG - send_gif = False and this message would send gif = False
2023-11-01 20:23:15,885 - octoprint.plugins.telegram.TMSG - DEBUG - Printer Status{'state': {'text': 'Operational', 'flags': {'operational': True, 'printing': False, 'cancelling': False, 'pausing': False, 'resuming': False, 'finishing': False, 'closedOrError': False, 'error': False, 'paused': False, 'ready': True, 'sdReady': False}, 'error': ''}, 'job': {'file': {'name': None, 'path': None, 'display': None, 'origin': None, 'size': None, 'date': None}, 'estimatedPrintTime': None, 'averagePrintTime': None, 'lastPrintTime': None, 'filament': None, 'user': None}, 'currentZ': None, 'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeLeftOrigin': None}, 'offsets': {}, 'resends': {'count': 0, 'transmitted': 209, 'ratio': 0}}
2023-11-01 20:23:15,887 - octoprint.plugins.telegram.TMSG - DEBUG - TEMPS - {'tool0': {'actual': 21.3, 'target': 0.0, 'offset': 0}, 'bed': {'actual': 20.7, 'target': 0.0, 'offset': 0}, 'chamber': {'actual': None, 'target': None, 'offset': 0}}
2023-11-01 20:23:15,892 - octoprint.plugins.telegram - DEBUG - get_current_layers api key=xxxxxxxxxxxxxxxxxx
2023-11-01 20:23:15,894 - octoprint.plugins.telegram.TMSG - DEBUG - Layers - None
2023-11-01 20:23:15,898 - octoprint.plugins.telegram.TMSG - DEBUG - VARS - {'self': <octoprint_telegram.telegramNotifications.TMSG object at 0xaf158910>, 'payload': {}, 'kwargs': {'chatID': '86204352', 'event': 'StatusNotPrinting', 'with_image': True, 'with_gif': 0, 'silent': False, 'thumbnail': None}, 'status': {'state': {'text': 'Operational', 'flags': {'operational': True, 'printing': False, 'cancelling': False, 'pausing': False, 'resuming': False, 'finishing': False, 'closedOrError': False, 'error': False, 'paused': False, 'ready': True, 'sdReady': False}, 'error': ''}, 'job': {'file': {'name': None, 'path': None, 'display': None, 'origin': None, 'size': None, 'date': None}, 'estimatedPrintTime': None, 'averagePrintTime': None, 'lastPrintTime': None, 'filament': None, 'user': None}, 'currentZ': None, 'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeLeftOrigin': None}, 'offsets': {}, 'resends': {'count': 0, 'transmitted': 209, 'ratio': 0}}, 'event': 'StatusNotPrinting', 'z': 0.0, 'temps': {'tool0': {'actual': 21.3, 'target': 0.0, 'offset': 0}, 'bed': {'actual': 20.7, 'target': 0.0, 'offset': 0}, 'chamber': {'actual': None, 'target': None, 'offset': 0}}, 'bed_temp': 20.7, 'bed_target': 0.0, 'e1_temp': 21.3, 'e1_target': 0.0, 'e2_temp': 0.0, 'e2_target': 0.0, 'percent': 0, 'Layers': None, 'currentLayer': '?', 'totalLayer': '?', 'time_done': '00:00:00', 'time_left': '[Unknown]', 'time_finish': '[Unknown]', 'file': None, 'path': None, 'owner': '', 'user': ''}
2023-11-01 20:23:15,902 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: Not printing.
Bed 20.7/0.0, Extruder 21.3/0.0.
2023-11-01 20:23:15,904 - octoprint.plugins.telegram.TMSG - DEBUG - {'chatID': '86204352', 'event': 'StatusNotPrinting', 'with_image': True, 'with_gif': 0, 'silent': False, 'thumbnail': None, 'markup': 'off', 'inline': False}
2023-11-01 20:23:15,908 - octoprint.plugins.telegram - DEBUG - start _send_msg
2023-11-01 20:23:15,910 - octoprint.plugins.telegram - DEBUG - PreImgMethod None
2023-11-01 20:23:15,917 - octoprint.plugins.telegram - DEBUG - log instead log sending message 
2023-11-01 20:23:15,920 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True, 'chat_id': '86204352', 'disable_notification': False}
2023-11-01 20:23:15,922 - octoprint.settings - WARNING - DeprecationWarning: Detected access to deprecated settings path ['webcam', 'snapshot'], returned value is derived from compatibility overlay. Please use the webcam system introduced with 1.9.0, this compatibility layer will be removed in a future release.
2023-11-01 20:23:15,925 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://127.0.0.1:1111/?action=snapshot
2023-11-01 20:23:16,141 - octoprint.plugins.telegram - DEBUG - Image transformations [H:None, V:None, R:None]
2023-11-01 20:23:16,142 - octoprint.plugins.telegram - DEBUG - message so far: Not printing.
Bed 20.7/0.0, Extruder 21.3/0.0.
2023-11-01 20:23:16,143 - octoprint.plugins.telegram - DEBUG - Sending with image.. 86204352
2023-11-01 20:23:17,341 - octoprint.plugins.telegram - DEBUG - Sending finished. 400
2023-11-01 20:23:17,344 - octoprint.plugins.telegram - DEBUG - PostImgMethod None
```
Not sure if it's relevant, but I had to compile mjpg-streamer myself due to the 32 bit architecture of my server. I got the mjpg-streamer from here: https://github.com/jacksonliam/mjpg-streamer

I installed the plugin directly from the plugin manager.

Any ideas on what is going wrong here? Let me know of any extra info I should include.

frosty656, 2023-11-05:
Also have the same issue.

pilotalex, 2023-11-08:
I also have the same issue. It started for me in September, it was working previously. May be related to the upgrade to OctoPrint 1.9.3

Hardbaer, 2023-11-09:
I also have the same issue.
It seems that it does not work with the changed camera stack in 1.9. Notifications without image work like before. As soon as an image is part of the notification it does not work anymore. The notification does not show up. Removing image in settings and the notification shows up again.

Hardbaer, 2023-11-09:
Came across #416. The solution solves this problem.

closed 2025-01-03 by giloser