jacopotediosi/OctoPrint-Telegram #422: No Multicam pictures

Vossberger, 2024-03-14:
Following #416 Version 1.6.6 Telegram plugin solved the issue not having any pictures anymore.

I'm missing pictures of my second Cam (multicam plugin fully configured), any idea how to get this working again? Seems like there are no updates on the Telegram plugin anymore ?

giloser, 2025-01-03:
Hi ,

could you try this version please?
https://github.com/giloser/OctoPrint-Telegram/archive/refs/tags/1.6.9.zip

jacopotediosi, 2025-06-24:
With the release of [version 1.7.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0), which includes a complete rewrite of the webcam management, I expect that multicam now works correctly.  
If that’s not the case, please open a new issue and attach the following:

- Details about your setup: operating system, OctoPrint version, plugin version
- Screenshot of your Classic Webcam and/or Multicam configurations
- The plugin logs, which can be found at *OctoPrint Settings → Logging → plugin_telegram.log*. If possible, please enable debug logging for the plugin as explained in [this wiki page](https://github.com/jacopotediosi/OctoPrint-Telegram/wiki/Where-are-the-logs%3F)

closed 2025-06-24 by jacopotediosi