jacopotediosi/OctoPrint-Telegram #424: upgrading to 1.10.0 - files and print command fail with list_files() got an unexpected keyword argument 'destinations'

guysoft, 2024-05-02:
Hey,
Recently upgraded to OctoPrint 1.10.0 , and now when I run ``/print`` or ``/files`` I get in Octoprint.log:
```
2024-05-02 20:38:35,252 - octoprint.plugins.telegram.TCMD - ERROR - An Exception in fileList : list_files() got an unexpected keyword argument 'destinations'
```
Thats it, no other error.
Telegram hangs on "loading files"

stedaho, 2024-05-02:
I can confirm that. Upgraded to OctoPrint 1.10.0 yesterday and now getting the same exception when trying to list files/start print. So no printing via telegram is possible at the moment! 
Telegram Notifications v 1.6.5

guysoft, 2024-05-02:
Ok, got a fix

guysoft, 2024-05-02:
https://github.com/fabianonline/OctoPrint-Telegram/pull/425/

You can install and test here:
https://github.com/guysoft/OctoPrint-Telegram/archive/refs/heads/bugfix/424.zip

stedaho, 2024-05-02:
Thanks a lot, I will try and give feedback here once my current long-running print has been finished!

stedaho, 2024-05-03:
I just tried your patch and I can confirm that it is working like a charm!

PR #425 (2024-05-05): Remove destinations keyword #424
Fixes #424, tested against latest OctoPrint, did not test against older versions 

You can test by installing:
https://github.com/guysoft/OctoPrint-Telegram/archive/refs/heads/bugfix/424.zip
closed 2024-05-05 by derpicknicker1