jacopotediosi/OctoPrint-Telegram #426: Update to 1.6.6 not possible

FindusD, 2024-05-06:
After running the update from 1.6.5 to 1.6.6 I always get the success message. 
But after restarting Ocoprint, I am prompted to update again each time. The plugin manager still shows version 1.6.5. as installed.

Investegate, 2024-05-06:
Same problem here. 
It seams there is still the old 1.6.5 version number in this build...

[image]

FiveTeethless, 2024-05-07:
same here.
Most probably this line is the reason:
https://github.com/fabianonline/OctoPrint-Telegram/blob/11047f9db53d720cd30e641488972d07564cd70a/setup.py#L17

jmarc67, 2024-05-08:
All is fine now with 1.6.7. Thanks

closed 2025-01-03 by giloser