jacopotediosi/OctoPrint-Telegram #428: Memory issue

AndreasAusD, 2024-05-09:
after updating the plugin to latest version 1.6.7 memory usage of octoprint is slowly increasing to max after i start a print.
after about five minutes the pi freezes and the print is interrupted.

normal print is possible when i disable the plugin.

setup : 
Raspberry Pi 3 Model B Plus Rev 1.3pi
Raspbian GNU/Linux 10 (buster) with latest updates
OctoPrint 1.10.0 Python 3.7.3 OctoPi 0.18.0

giloser, 2024-06-10:
could you give us some more information?
what is your configuration for message (with, without images, gif, ...)
did it work correctly before the update? did you change your configuration between?

thanks

jacopotediosi, 2025-07-11:
Cannot replicate, no logs, no user feedback

closed 2025-07-11 by jacopotediosi