jacopotediosi/OctoPrint-Telegram #432: ERR GET IMAGE with WebCam

cankurall, 2024-11-11:
Hello, I'm running Octo4a on a tablet. I’ve directed my computer's camera over IP using the Yawcam application. There’s no issue with live or snapshot images on Octoprint, and I can also view them in my browser. However, the Telegram bot is not transmitting any images and instead sends the message '[ERR GET IMAGE]'. Could you please help me with this?

[image]
[image]
[image]

Telegram:
[image]

tsujovolsky, 2024-11-21:
I'm getting the same errors using octoprint for windows and yawcam. In addition to what @cankurall is getting, here are also the logs:

> 2024-11-21 10:00:07,429 - octoprint.settings - WARNING - DeprecationWarning: Detected access to deprecated settings path ['webcam', 'snapshot'], returned value is derived from compatibility overlay. Please use the webcam system introduced with 1.9.0, this compatibility layer will be removed in a future release.
> 2024-11-21 10:00:07,430 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://192.168.1.11:1111/api/frame.jpeg?src=webcam1
> 2024-11-21 10:00:07,560 - octoprint.plugins.telegram - DEBUG - Image transformations [H:None, V:None, R:None]
> 2024-11-21 10:00:07,561 - octoprint.plugins.telegram - DEBUG - message so far: [ERR GET IMAGE]
> 
>  
> 2024-11-21 10:00:07,561 - octoprint.plugins.telegram - DEBUG - Sending without image.. 605302509

It takes my snapshot url as `http://192.168.1.11:1111/api/frame.jpeg?src=webcam1` whereas the actual one is `http://192.168.1.11:8888/out.jpg?action=snapshot `

giloser, 2025-01-03:
Hi ,

could you try this version please?
https://github.com/giloser/OctoPrint-Telegram/archive/refs/tags/1.6.9.zip

gDanix, 2025-01-16:
Hi! I've installed Octoprint today and stumbled upon this problem today.

Nothing worked, and I thought to myself: "Heck, Octoprint is returning invalid URLs. This is too much to debug, let's hardcode the appropriate URL and call it the day"

The code (which is inserted in the place of [this line that sets the URL](https://github.com/giloser/OctoPrint-Telegram/blob/40ebbad003e851c88599ab6d4df3fd457a95373f/octoprint_telegram/__init__.py#L2034)):
```
self._logger.debug("get_webcam_configurations : " + str(config))
self._logger.debug("Camera URL hardcoded to http://10.100.100.5:8082/snapshot")
#url = config.snapshotDisplay
url = "http://10.100.100.5:8082/snapshot"
self._logger.debug("cam conf URL : " + str(url))
```

The resulting log messages:
```
2025-01-16 22:11:07,679 - octoprint.plugins.telegram - DEBUG - self.snap_new_method : True
2025-01-16 22:11:07,681 - octoprint.plugins.telegram - DEBUG - cameras {'classic': <octoprint.webcams.ProvidedWebcam object at 0xffff5c0279d0>}
2025-01-16 22:11:07,683 - octoprint.plugins.telegram - DEBUG - camera : classic
2025-01-16 22:11:07,686 - octoprint.plugins.telegram - DEBUG - get_webcam_configurations : name='classic' displayName='Classic Webcam' canSnapshot=True snapshotDisplay='http://10.100.100.1:1111/snapshot' flipH=False flipV=False rotate90=False extras={'stream': 'http://10.100.100.1:1111/stream', 'streamTimeout': 5, 'streamRatio': <RatioEnum.sixteen_nine: '11:1'>, 'streamWebrtcIceServers': ['stun:stun.l.google.com:19302'], 'cacheBuster': False} compat=WebcamCompatibility(streamTimeout=5, streamRatio='11:1', streamWebrtcIceServers=['stun:stun.l.google.com:19302'], cacheBuster=False, stream='http://10.100.100.1:1111/stream', snapshot='http://10.100.100.1:1111/snapshot', snapshotTimeout=5, snapshotSslValidation=True)
2025-01-16 22:11:07,689 - octoprint.plugins.telegram - DEBUG - Camera URL hardcoded to http://10.100.100.1:1111/snapshot
2025-01-16 22:11:07,692 - octoprint.plugins.telegram - DEBUG - cam conf URL : http://10.100.100.1:1111/snapshot
2025-01-16 22:11:07,695 - octoprint.plugins.telegram - DEBUG - will try to get the snapshot from the plugin for classic
2025-01-16 22:11:07,727 - octoprint.plugins.telegram - DEBUG - set the snapshot to data to try to continue
```

I don't how on earth can a hardcoded value be changed after an installation process, if someone is responsible for this and (if someone is) in what was thinking (if at all). I'm still trying (and failing) to process what I've just seen.

Good luck, @giloser trying to fix this. I'm SO done with Python and its JS-level bs...

giloser, 2025-01-17:
Hi, 

for the URL of the camera I have similar change in my log but it doesn't seems to be a problem for me...

do you use the new camera stack? 

what do you mean by nothing worked? you got message? but no image?

gDanix, 2025-01-17:
Hi!

> do you use the new camera stack?

To be precise, I checked out the code from the Pull Request #430, and then added the hardcoded URL. I think that Pull Request is your work moving the project over to the new camera stack, but I didn't investigate further.

> what do you mean by nothing worked? you got message? but no image?
Yes, exactly, I got the message, but instead of the image, I got the [ERR GET IMAGE] text prepended to the message.

Just for your information, in my setup there exists a 10.100.100.1 host, but there's no software in it listening from port 1111. It seems like [the "something" that is magically changing hardcoded strings] is setting to 1 the last digit of the last number of the IP (this is probably why @tsujovolsky didn't reported any issues with the IP, since their IP already ends in 1), and changing the port to 1111.

I've not done experiments trying to obfuscate the hardcoded IP, to see if it's an installation thing that is statically checking the code, or a runtime thing that is constantly checking strings, I've just uninstalled the plugin, sorry.

I'm running OctoPrint 1.10.3 with Python 3.9.2.

giloser, 2025-01-17:
Ok thanks for the answer

I think that PR was a first try but was still not working that why I ask to try with the link to my full version. If people confirm it's working then I'll try to create a new PR and see if we can merge it to current version. 
If not I'll just continue on my version for me.
here is a link to the version where the gif should work now.

https://github.com/giloser/OctoPrint-Telegram/archive/refs/tags/1.6.92.zip

So please if you want to use the plugin try this version and let me know.

gDanix, 2025-01-17:
I'm sorry but I wasted too much hours yesterday and after setting everything my SBC and my instance is very slow. At this point I would look for rewrite parts of OctoPrint in a more appropriate language rather than spending more time with Python magic... Sorry

jacopotediosi, 2025-07-19:
Regarding above users’ concerns about incorrect URLs being logged: in older versions of OctoPrint-Telegram, a filter was mistakenly altering the logs. The URLs users configured were actually the ones being used, even if the logs showed incorrect values. The actual bug for the ERR_GET_IMAGE error is likely something else — such as the absence of `cpulimit` or a failure in `ffmpeg`. I will verify the correct behavior of the plugin on Octo4A and post updates in issue #387.