jacopotediosi/OctoPrint-Telegram #433: [ERR GET IMAGE] - SSL problem?

SM-26, 2025-01-05:
Hello,

long time user of this plugin. great stuff.

OctoPrint v1.10.3 running on a Pi4 with OctoPi 0.18.0
Multicam at 0.4.2, getting both streams just fine on the webui.

 I've tried: latest, old-master, @giloser 1.6.9

I know I'm stupid somehow, just not sure who yet.
Thanks in advance.

``` log
2025-01-05 22:26:10,600 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: Not printing.  \n Bed 19.88/0.0, Extruder 17.81/0.0.
2025-01-05 22:26:10,600 - octoprint.plugins.telegram.TMSG - DEBUG - {'chatID': '278683632', 'event': 'StatusNotPrinting', 'with_image': True, 'with_gif': 0, 'silent': False, 'thumbnail': None, 'markup': 'off', 'inline': False}
2025-01-05 22:26:10,602 - octoprint.plugins.telegram - DEBUG - start _send_msg
2025-01-05 22:26:10,603 - octoprint.plugins.telegram - DEBUG - PreImgMethod None
2025-01-05 22:26:10,607 - octoprint.plugins.telegram - DEBUG - log instead log sending message 
2025-01-05 22:26:10,609 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True, 'chat_id': '278683632', 'disable_notification': False}
2025-01-05 22:26:10,610 - octoprint.plugins.telegram - DEBUG - self.snap_new_method : True
2025-01-05 22:26:10,611 - octoprint.plugins.telegram - DEBUG - cameras {'multicam/Pi Cam': <octoprint.webcams.ProvidedWebcam object at 0xa869bc10>, 'multicam/Nozzle Cam': <octoprint.webcams.ProvidedWebcam object at 0xa869b270>}
2025-01-05 22:26:10,612 - octoprint.plugins.telegram - DEBUG - camera : multicam/Pi Cam
2025-01-05 22:26:10,613 - octoprint.plugins.telegram - DEBUG - get_webcam_configurations : name='multicam/Pi Cam' displayName='Pi Cam' canSnapshot=True snapshotDisplay='https://192.168.1.228/webcam/?action=snapshot' flipH=False flipV=False rotate90=False extras={'stream': 'https://192.168.1.228/webcam/?action=stream', 'streamTimeout': 15, 'streamRatio': '11:1', 'cacheBuster': True} compat=WebcamCompatibility(streamTimeout=15, streamRatio='11:1', streamWebrtcIceServers=[], cacheBuster=True, stream='https://192.168.1.228/webcam/?action=stream', snapshot='https://192.168.1.228/webcam/?action=snapshot', snapshotTimeout=15, snapshotSslValidation=True)
2025-01-05 22:26:10,614 - octoprint.plugins.telegram - DEBUG - cam conf URL : https://192.168.1.228/webcam/?action=snapshot
2025-01-05 22:26:10,615 - octoprint.plugins.telegram - DEBUG - will try to get the snapshot from the plugin for multicam/Pi Cam
2025-01-05 22:26:10,664 - octoprint.plugins.telegram - ERROR - Exception get snaposhot from octoprint: HTTPSConnectionPool(host='192.168.1.228', port=443): Max retries exceeded with url: /webcam/?action=snapshot (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1056)')))
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 467, in _make_request
    self._validate_conn(conn)
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 1092, in _validate_conn
    conn.connect()
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connection.py", line 651, in connect
    assert_fingerprint=self.assert_fingerprint,
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connection.py", line 786, in _ssl_wrap_socket_and_match_hostname
    tls_in_tls=tls_in_tls,
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/util/ssl_.py", line 466, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls, server_hostname)
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/util/ssl_.py", line 510, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.7/ssl.py", line 412, in wrap_socket
    session=session
  File "/usr/lib/python3.7/ssl.py", line 886, in _create
    self.do_handshake()
  File "/usr/lib/python3.7/ssl.py", line 1150, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1056)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 802, in urlopen
    **response_kw,
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 491, in _make_request
    raise new_e
urllib3.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1056)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/adapters.py", line 497, in send
    chunked=chunked,
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 845, in urlopen
    method, url, error=new_e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/util/retry.py", line 515, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.1.228', port=443): Max retries exceeded with url: /webcam/?action=snapshot (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1056)')))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... 65 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 2770, in take_image
    r = requests.get(snapshot_url, timeout=10, proxies=self.getProxies())
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/adapters.py", line 517, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='192.168.1.228', port=443): Max retries exceeded with url: /webcam/?action=snapshot (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1056)')))
2025-01-05 22:26:10,924 - octoprint.plugins.telegram - DEBUG - take_image return data : None
2025-01-05 22:26:10,925 - octoprint.plugins.telegram - DEBUG - camera : multicam/Nozzle Cam
2025-01-05 22:26:10,925 - octoprint.plugins.telegram - DEBUG - get_webcam_configurations : name='multicam/Nozzle Cam' displayName='Nozzle Cam' canSnapshot=True snapshotDisplay='https://192.168.1.228/webcam2/?action=snapshot' flipH=False flipV=False rotate90=False extras={'stream': 'https://192.168.1.228/webcam2/?action=stream', 'streamTimeout': 15, 'streamRatio': '11:1', 'cacheBuster': True} compat=WebcamCompatibility(streamTimeout=15, streamRatio='11:1', streamWebrtcIceServers=[], cacheBuster=True, stream='https://192.168.1.228/webcam2/?action=stream', snapshot='https://192.168.1.228/webcam2/?action=snapshot', snapshotTimeout=15, snapshotSslValidation=True)
2025-01-05 22:26:10,925 - octoprint.plugins.telegram - DEBUG - cam conf URL : https://192.168.1.228/webcam2/?action=snapshot
2025-01-05 22:26:10,926 - octoprint.plugins.telegram - DEBUG - will try to get the snapshot from the plugin for multicam/Nozzle Cam
2025-01-05 22:26:10,979 - octoprint.plugins.telegram - ERROR - Exception get snaposhot from octoprint: HTTPSConnectionPool(host='192.168.1.228', port=443): Max retries exceeded with url: /webcam2/?action=snapshot (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1056)')))

```

giloser, 2025-01-17:
Hi, thanks for the info.

I would say that the problem is a security problem you set a self signed certificate and when we try to get the image it check the certificate and say it's not valid. 

There is probably a setting to set on the request to not check the certificate but since we try to let the camera designed plugin to take the image and send it to us , I think you should ask on the multicam plugin if he can made something about that.

Sorry

giloser, 2025-01-22:
I don't know, I don't see anything else in the log that would explain it.

closed 2025-06-24 by jacopotediosi