jacopotediosi/OctoPrint-Telegram #442: [Bug] Shutdown message not sent

spanzetta, 2025-07-12:
### A friendly checklist before filing your bug report

- [x] I have verified that the bug concerns or is caused by OctoPrint-Telegram
- [x] I have verified that the bug occurs with the latest version of OctoPrint-Telegram
- [x] I have verified that there isn't already an open issue for this bug

### Bug description

Enable Shutdown command in Telegram plugin 
Run Shutdown command from OctoPrint 
There is no message sent

### Version of OctoPrint

1.11.2

### Operating system running OctoPrint

OctoPi 1.0

### Version of OctoPrint-Telegram

1.7

### Make sure to attach the following files in the textarea below. Please be aware that issues without logs will be closed without a response.

- [x] OctoPrint Systeminfo Bundle. See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [x] Plugin logs, preferably captured with debug-level logging enabled. See [here](https://github.com/jacopotediosi/OctoPrint-Telegram/wiki/Where-are-the-logs%3F) if you don't know where to find them.
- [ ] JavaScript console logs from your browser. See [here](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers) if you don't know where to find them.
- [ ] Screenshots and/or videos showing the problem.

### Additional information & file uploads

[octoprint-systeminfo-20250712101009.zip](https://github.com/user-attachments/files/21194783/octoprint-systeminfo-20250712101009.zip)

### Final checklist

- [x] The title I chose for this issue clearly describes the bug (no "please help" or "it doesn't work") and is under 100 characters
- [x] This report is about a single bug only (if you encounter more than one bug, open a separate ticket for each)

jacopotediosi, 2025-07-12:
@spanzetta Plugin logs are missing, you attached only systeminfo bundle.

spanzetta, 2025-07-12:
Ops.. sorry.. here it is..

[plugin_telegram.log.txt](https://github.com/user-attachments/files/21194827/plugin_telegram.log.txt)

spanzetta, 2025-07-12:
Thanks to you!

At least you know about the bug..

I will patiently wait for the fix

Thanks for the very fast analysis

spanzetta, 2025-07-17:
> Ultimately, it appears this is a bug in OctoPrint. Reference: [OctoPrint/OctoPrint#5176](https://github.com/OctoPrint/OctoPrint/issues/5176)

Interesting.. thanks for the info

spanzetta, 2025-07-17:
Thanks.. will try tomorrow as currently my printer is busy with a long print..

spanzetta, 2025-07-18:
When I try to edit octoprint.service (using vi) it says it's read-only file..
Can you pls guide me how to modify it properly (eg. should I stop octoprint service first?)
Thanks

spanzetta, 2025-07-18:
Thanks for the hint
Will try and let you know

spanzetta, 2025-07-18:
It seems it works now..
Also when running the sudo systemctl restart octoprint command I got the "Shutting down. Goodbye" message..
Thanks a lot for the quick fix!