jacopotediosi/OctoPrint-Telegram #444: [Bug/Feature] PrintStart notification sends multiple images, making (iOS) notifications just say "2 images"

melyux, 2025-07-19:
### The problem

This is an annoying thing with Telegram itself where if you send multiple media in the same message, the iOS notifications simply say "2 images" or "3 images". This is the case with the PrintStart message, which includes both an image of the model and a snapshot from the webcam. Unfortunately, this message is also the one that says the ETA and print time and what's printing, so it's especially important to have these actually show in the Notification Center instead of just saying "2 images".

### Proposed solution

Maybe these PrintStart messages should be split into 2 messages, or the snapshot image made optional?

### Additional context

Love you

jacopotediosi, 2025-07-19:
Uhm, this use case seems particularly specific. There are definitely multiple possible solutions to consider, but I’m just curious — @melyux, are you sure the best option for your case isn’t simply disabling photos and videos for the PrintStart notification message?

I’m also not entirely clear on why you want the ETA to be shown in the notification bar, since I assume the next notification would overwrite it anyway.

melyux, 2025-07-19:
It doesn't overwrite it but they stack, so you can see everything without having to open the app. I do want to see the model's thumbnail though, but I could do without the webcam snapshot of the printer... maybe an option to turn off the printer webcam snapshot for this PrintStart message?

jacopotediosi, 2025-07-20:
Labelling as done since a solution has been proposed.  
The issue will be closed once the user confirms, or otherwise in a few days.  
If the solution turns out to be unsatisfactory, I'm open to considering alternatives.

melyux, 2025-07-20:
Acceptable solution once I updated to 1.8.0 and saw it. Thanks