jacopotediosi/OctoPrint-Telegram #449: [Bug] 1.9.0 failing to load after updating from 1.8.0

sanstey, 2025-08-25:
### A friendly checklist before filing your bug report

- [x] I have verified that the bug concerns or is caused by OctoPrint-Telegram
- [x] I have verified that the bug occurs with the latest version of OctoPrint-Telegram
- [x] I have verified that there isn't already an open issue for this bug

### Bug description

After installing the 1.9.0 update, the plugin management log popup showed "Install Unknown" and octoprint.log reports the following:
`2025-08-25 03:34:32,194 - octoprint.plugin.core - ERROR - Error loading plugin telegram
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1339, in _import_plugin
    module = _load_module(spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 74, in _load_module
    spec.loader.exec_module(sys.modules[spec.name])
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 31, in <module>
    from .telegram_commands import TCMD
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/telegram_commands.py", line 30, in <module>
    class TCMD:
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/telegram_commands.py", line 923, in TCMD
    def split_with_escape_handling(self, param_str: str, separator: str) -> list[str]:
TypeError: 'type' object is not subscriptable`

### Version of OctoPrint

1.11.2

### Operating system running OctoPrint

OctoPi 0.17.0

### Version of OctoPrint-Telegram

1.9.0

### Make sure to attach the following files in the textarea below. Please be aware that issues without logs will be closed without a response.

- [x] OctoPrint Systeminfo Bundle. See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [x] Plugin logs, preferably captured with debug-level logging enabled. See [here](https://github.com/jacopotediosi/OctoPrint-Telegram/wiki/Where-are-the-logs%3F) if you don't know where to find them.
- [ ] JavaScript console logs from your browser. See [here](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers) if you don't know where to find them.
- [ ] Screenshots and/or videos showing the problem.

### Additional information & file uploads

[octoprint-systeminfo-20250825034910.zip](https://github.com/user-attachments/files/21960636/octoprint-systeminfo-20250825034910.zip)

[plugin_telegram.log](https://github.com/user-attachments/files/21960638/plugin_telegram.log)

### Final checklist

- [x] The title I chose for this issue clearly describes the bug (no "please help" or "it doesn't work") and is under 100 characters
- [x] This report is about a single bug only (if you encounter more than one bug, open a separate ticket for each)

jacopotediosi, 2025-08-25:
@sanstey, Could you please install the plugin by entering the following URL into the Plugin Manager:
```
https://github.com/jacopotediosi/OctoPrint-Telegram/archive/bugfix.zip
```
And let me know if it works for you?

sanstey, 2025-08-25:
@jacopotediosi, the plugin manager log popup still showed _Unknown_, but otherwise everything seems to be working fine!

jacopotediosi, 2025-08-25:
I’ve just released [version 1.9.1](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.9.1)
 to fix this bug for everyone. Thank you very much for reporting it.

commit af2d5a25 (2025-08-25): Fix "TypeError: 'type' object is not subscriptable"

This should fix compatibility for python<3.9
Fix #449